grisp/grisp #2: grisp_led is accepting interval 0 (as well as negative intervals)

ThomasArts, 2017-07-21:
Nobody in its right mind is ever to flash a led with interval 0 or -1, but there is a subtle difference between the two.

When using a negative interval, the grisp_led server will crash (due to a call to send_after with negative number). It will be restarted and the leds will all be black.

When using a zero interval, the led will constantly be switched on an off in the rate the hardware can handle. This may eventually fill the mail box.

nextl00p, 2017-07-25:
What should be the intended behaviour? Turn it off?

At the moment the interval is given as-is to the timer which expects a value >= 0.

We cannot use throw, to indicate failure to the caller, as everything is happening in a async cast, so simply turning it off is the best idea in my opinion.

eproxus, 2017-08-08:
I think it would be confusing to implement different behaviors just based on the integers. I'd prefer is function clause errors would be generated for anything below 1.

ThomasArts, 2017-08-09:
Agreed, having a function clause error makes full sense.

closed 2017-08-23 by eproxus