grisp/grisp #11: PmodNAV magnetometer fails initialization "sometimes"

eproxus, 2017-11-07:
Tasks marked with ❌ fails:

1. ❌ Start PmodNAV driver when Pmod is inserted
2. Start PmodNAV driver again, leaving the Pmod inserted
3. Stop PmodNAV driver, leaving the Pmod inserted
4. ❌ Start PmodNAV driver again, leaving the Pmod inserted
5. ❌ Start PmodNAV driver again, leaving the Pmod inserted
6. ❌ [repeatedly fails...]
7. Remove Pmod physically
8. Inserting the Pmod
9. Start PmodNAV driver, leaving the Pmod inserted
8. Stop PmodNAV driver, leaving the Pmod inserted
10. ❌Start PmodNAV driver again, leaving the Pmod inserted
11. ❌ [repeatedly fails...]

Example:

```
1> grisp:add_device(spi1, pmod_nav).
** exception error: no match of right hand side value
                    {error,
                        {{{register_mismatch,acc,who_am_i,<<129>>},
                          [{pmod_nav,verify_reg,2,
                               [{file,
                                    "_checkouts/grisp/src/pmod_nav.erl"},
                                {line,128}]},
                           {lists,foldl,3,[{file,"lists.erl"},{line,1263}]},
                           {pmod_nav,init,1,
                               [{file,
                                    "_checkouts/grisp/src/pmod_nav.erl"},
                                {line,52}]},
                           {gen_server,init_it,6,
                               [{file,"gen_server.erl"},{line,328}]},
                           {proc_lib,init_p_do_apply,3,
                               [{file,"proc_lib.erl"},{line,247}]}]},
                         {child,undefined,spi1,
                             {pmod_nav,start_link,[spi1,#{}]},
                             permanent,5000,worker,
                             [pmod_nav]}}}
     in function  grisp_devices_sup:start_child/3 (_checkouts/grisp/src/grisp_devices_sup.erl, line 23)
     in call from grisp_devices:add_device/3 (_checkouts/grisp/src/grisp_devices.erl, line 37)
2> grisp:add_device(spi1, pmod_nav).
{device,spi1,pmod_nav,<0.114.0>,#Ref<0.0.1.3663>}
3> grisp:remove_device(grisp:device(spi1)).
ok
4> grisp:add_device(spi1, pmod_nav).
** exception error: no match of right hand side value
                    {error,
                        {{{register_mismatch,mag,who_am_i_m,<<"ÿ">>},
                          [{pmod_nav,verify_reg,2,
                               [{file,
                                    "_checkouts/grisp/src/pmod_nav.erl"},
                                {line,128}]},
                           {lists,foldl,3,[{file,"lists.erl"},{line,1263}]},
                           {pmod_nav,init,1,
                               [{file,
                                    "_checkouts/grisp/src/pmod_nav.erl"},
                                {line,52}]},
                           {gen_server,init_it,6,
                               [{file,"gen_server.erl"},{line,328}]},
                           {proc_lib,init_p_do_apply,3,
                               [{file,"proc_lib.erl"},{line,247}]}]},
                         {child,undefined,spi1,
                             {pmod_nav,start_link,[spi1,#{}]},
                             permanent,5000,worker,
                             [pmod_nav]}}}
     in function  grisp_devices_sup:start_child/3 (_checkouts/grisp/src/grisp_devices_sup.erl, line 23)
     in call from grisp_devices:add_device/3 (_checkouts/grisp/src/grisp_devices.erl, line 37)
```

nextl00p, 2017-11-07:
That's also my perceived erroneous behaviour.

eproxus, 2017-11-07:
After some fixes (among others actually pulling the pins) I now have the same problem with the altimeter instead of magnetometer, so some progress...

eproxus, 2018-07-30:
Fixed in 421385a9912cb9a9eef3adb29bb77601fb42e972.

closed 2018-07-30 by eproxus