grisp/grisp #16: Ubuntu 16.04 is missing build-essential package as dependency

jacob-s-son, 2017-12-08:
### Versions

* OS: `16.04`
* GRiSP Toolchain: `1726382032480d97fd0f2110c21a59fd08afa05c`
* Erlang: `19.3`

### Description

I was installing GRiSP tools on the fresh Ubuntu 16.04 LTS VM and had this step fail:
`asdf install erlang 19.3`

Basically it complained not being able to find GNU make. Quick search on the internet has shown that `build-essential` package is missing. After installing it, I could continue installation.

eproxus, 2017-12-11:
Thanks! I added the missing build dependencies to the wiki.

If I remember correctly Ubuntu ships with Python 3 by default, hence the `python27` name. Not sure what exactly what in the build process needs `python` to be Python 2.7
@jacob-s-son Are you referring to the building of the toolchain here?

jacob-s-son, 2017-12-11:
Thanks for the response, what I meant was - at some point there was an error when building the toolchain. Don't have the exact error message, but the gist of it was that `python` command was not found in the path.

I did some digging and apparently I had both installed  2.7 and 3.5, so in my `/usr/bin` directory there were two binaries `python2.7` and `python3.5` (or `python27` and `python35`), but none of them was symlinked as `python`.

The fix was rather clear - either symlink `python2.7` or `python3.5` as `python`. I had no idea which one to symlink as I wasn't aware what python version that script is compatible with.
I just did an educated guess and symlinked `2.7`, because in dependency list there is `python2.7-dev` listed as dependency.

I hope my comment makes a clearer explanation, what problems I had with `python` dependency.

eproxus, 2017-12-11:
Alright, so according to [this page](https://wiki.ubuntu.com/Python/3):

> What this **does not** mean: `/usr/bin/python` will point to Python 3. **No, this is not going to happen** (unless PEP 394 advocates otherwise, which is doubtful for the foreseeable future). `/usr/bin/python` and `/usr/bin/python2` will point to Python 2.7 and `/usr/bin/python3` will point to the latest supported Python 3 version.

On Ubuntu 17.10 installing `python-minimal` gets us the `/usr/bin/python` binary. Would it be enough to add this to the build dependencies?

eproxus, 2017-12-11:
I just tried this on Ubuntu Server 16.04.3 and installing `python-minimal` sets `/usr/bin/python` correctly. Have you installed that package or did you install Python 2.7 in some other way?

jacob-s-son, 2017-12-11:
Besides `build-essential` and `unzip` packages, I've only installed those that were listed as dependencies. Maybe, you are right and we should add `python-minimal` to the list

nextl00p, 2017-12-11:
Adding it won't hurt.

I tested with Ubuntu 16.04 Desktop. Ubuntu Server seems to include much less packages by default.

jacob-s-son, 2017-12-11:
:+1: I think it's enough for now to get this issue resolved, feel free to close it. 

I will try next weekend to finalise all my notes into a guide, how to get a working VM for developing GRiSP applications.

eproxus, 2017-12-11:
It wasn't included in desktop Ubuntu 17.10 either. I've added it here: https://github.com/grisp/grisp/wiki/Linux-Development-Environment Any other place?

jacob-s-son, 2017-12-11:
@eproxus I think it should do the work. One thing though - I think it's added twice in the list

eproxus, 2017-12-11:
Yeah, @nextl00p already added it (which I missed) 😄

closed 2017-12-11 by eproxus