grisp/grisp #30: grisp_i2c breaks when a longer message is sent after a short one

lwehmeier, 2018-06-15:
### Versions

* OS: Ubuntu 16.04
* Erlang: Erlang/OTP 20 Erts 9.2
* Rebar: rebar 3.5.3 on Erlang/OTP 20 Erts 9.2

### Description
When a short message is sent before a longer message, the grisp_i2c module only transmits $SHORTMSGLENGTH bytes of the longer message correctly and 0x00 after that.

Reproduce:

* `grisp_i2c:msgs([16#40, {write, <<16#23, 16#24, 16#25>>}]).` works fine
* `grisp_i2c:msgs([16#40, {write, <<16#42>>}, {write, <<16#23, 16#24, 16#25>>}]).` What happens on the I2C bus: `I2C_START to Address 40`, `I2C_WRITE`, `0x23`, `0x00`, `0x00`

[image]

peerst, 2018-06-16:
Since you already have a debugger connected:

Did you check if the msgs that are passed to the ioctl are correct or if the problem is visible already here:

https://github.com/grisp/grisp/blob/master/grisp/grisp_base/drivers/grisp_i2c_drv.c#L155

peerst, 2018-06-16:
BTW are you using OpenOCD for Debugging or did you connect a external JTAG debugger?

Be aware of this issue in OpenOCD:

https://github.com/grisp/grisp-software/issues/33

The description is a bit muddled (some of whats observed as "broken" is the normal "fun" of lowlevel JTAG debugging when the debugger doesn't know about ISRs and other Threads)

But what remains is that asm level single step has been broken in OpenOCD for our CPU.  Which basically prevents all kinds of single stepping or continue from a breakpoint.  Maybe its fixed already?  Does it work for you?

Whats supposed to work is run to a first Breakpoint and examine the State, but no continue or single step afterwards (looks as if stuck on a breakpoint).  Sounds very limiting but often the state of a program or the debuggers has been compromised after a dynamic system is stopped anyway.

lwehmeier, 2018-06-18:
Re. debugger: using a J-Link with Segger's gdb server, just tested single stepping, seems to work fine. I guess they implemented some kind of workaround for the caching issues.

Re. debugging: I still have to setup the grisp build environment to get a symbol table to debug properly. Need to figure out some build issues on windows first (the Segger software's not great on linux), I'll probably do that later today and update this post with the debugged ioctl calls.

Re. I2C: As it says in my edit, the problem mysteriously disappeared once I tried to debug it and I'm currently no longer able to reproduce the problem. Even though I did not modify the flash or SD card contents..

I have just run into another i2c issue that might be related: `grisp_i2c:msgs([16#40, {write, <<16#80>>}, {write, <<16#55, 16#AA, 16#55>>}, {write, <<32>>}, {write, <<0:40>>}]).` causes grisp to crash:
```
assertion "(address & SEVEN_BIT_MASK) == 0" failed: file "/Users/kilian/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c", line 152, function: atsam_i2c_set_address_regs
fatal extension: source=7, is_internal=0, error=541069544
```

`grisp_i2c:msgs([16#40, {write, <<16#80>>}, {write, <<16#55, 16#AA, 16#55>>}, {write, <<0:40>>}]).` results in a crash with the following message
```
assertion "(address & SEVEN_BIT_MASK) == 0" failed: file "/Users/kilian/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c", line 152, function: atsam_i2c_set_address_regs
fatal extension: source=7, is_internal=0, error=541069544
fatal extension: source=9, is_internal=0, error=541069464
R0   = 0x40018000 R8  = 0x00000005
R1   = 0x00000100 R9  = 0x00000000
R2   = 0x00000078 R10 = 0x00000000
R3   = 0x0200A02A R11 = 0x00000000
R4   = 0x709C6D10 R12 = 0x00000000
R5   = 0x40018000 SP  = 0x00000080
R6   = 0x00000001 LR  = 0x70326569
R7   = 0x00005B20 PC  = 0x7032656C
XPSR = 0x41000023 VEC = 0x00000005
```
Note that the same assertion error once includes a register dump and once does not. That's not a c&p error.
Both commands produce activity on the i2c bus right before the driver crashes:
[image]
[image]

peerst, 2018-06-18:
Re Debugging:  There are two ways to attach a debugger to the GRiSP board:

1. Just a USB Cable via the on board USB->JTAG support and just a software solution (OpenOCD) on the host.  This has the single stepping bug

2. Via a JTAG debug dongle and the JTAG connector on the board: this has no such issue with single stepping

You are using #2 therefore it works.

Th latter behaviour that you observed could be caused by the same messed up msgs for the ioctl.  Apparently it comes from the lowlevel i2c driver complaining about a adr not being 7 bit

lwehmeier, 2018-06-19:
So, for `grisp_i2c:msgs([16#40, {write, <<16#80>>}, {write, <<16#55, 16#AA, 16#55>>}, {write, <<0:40>>}]).` the ioctl call `rv = ioctl(grisp_i2c_data.fd, I2C_RDWR, &msgset);` is made for `fd=25`, which is the same filedescriptor used in i2c calls that work ok. `msgset` contains valid pointers to three message structs:
```
  msgs = 0x70a8b194,
  nmsgs = 3
```
`msgset.msgs[0]`:
```
  addr = 64,
  flags = 0,
  len = 1,
  buf = 0x7368004a "\200U\252U"
```
`msgset.msgs[1]`:
```
  addr = 64,
  flags = 0,
  len = 3,
  buf = 0x7368004b "U\252U"
```
`msgset.msgs[2]`:
```
  addr = 64,
  flags = 0,
  len = 5,
  buf = 0x7368004e ""
```
So the ioctl call provides correct message lengths and addresses.
After some transmitting several bytes, the assertion in atsam_i2c_set_address_regs fails for no reason:
```
(gdb) c
Continuing.
Breakpoint 9, atsam_i2c_set_address_regs (regs=0x40018000, address=64, ten_bit_addr=false, read_transfer=false) at /home/leon/dev/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bu
s.c:152
152                     assert((address & SEVEN_BIT_MASK) == 0);
(gdb) info args
regs = 0x40018000
address = 64
ten_bit_addr = false
read_transfer = false
(gdb) delete breakpoints
(gdb) break /home/leon/dev/msc/grisp-software/rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c:153
Breakpoint 10 at 0x703269c4: file /home/leon/dev/grisp-software/build/..//rtems/c/src/lib/libbsp/arm
/atsam/i2c/atsam_i2c_bus.c, line 153.
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
atsam_i2c_set_address_regs (regs=0x40018000, address=64, ten_bit_addr=false, read_transfer=false) at /home/leon/dev/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c:152
152                     assert((address & SEVEN_BIT_MASK) == 0);
(gdb) info args
regs = 0x40018000
address = 64
ten_bit_addr = false
read_transfer = false
(gdb) print address
$2 = 64
(gdb) print 64&0xFF80
$3 = 0
```

sebhub, 2018-06-21:
Could you please set a break point to "break __assert_func" and then also print out a stack trace (GDB "bt" command) if the assertion triggers.

c-mauderer, 2018-06-21:
> Note that the same assertion error once includes a register dump and once does not. That's not a c&p error.

Just a note regarding that: There is a small difference between the two. The first message only has a fatal with source=7. That is a `RTEMS_FATAL_SOURCE_ASSERT` which is to be expected if an assert went wrong (also it's odd that the assert went wrong). The second message has an assert and a second fatal with source=9. 9 is a `RTEMS_FATAL_SOURCE_EXCEPTION` which should happen only on unhandled exceptions. It's quite hard to tell why this one happened.

lwehmeier, 2018-06-22:
`grisp_i2c:msgs([16#40, {write, <<16#80>>}, {write, <<16#55, 16#AA, 16#55>>}, {write, <<32>>}, {write, <<0:40>>}]).`:
```
Breakpoint 2, __assert_func (file=file@entry=0x7045adbc "/home/leon/dev/msc/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c", line=line@entry=152, func=func@entry=0x7045ad68 <__func__.15959> "atsam_i2c_set_address_regs", failedexpr=failedexpr@entry=0x7045ae38 "(address & SEVEN_BIT_MASK) == 0") at /home/leon/dev/msc/grisp-software/build/..//rtems/c/src/../../cpukit/libcsupport/src/__assert.c:37

(gdb) bt
#0  __assert_func (file=file@entry=0x7045adbc "/home/leon/dev/msc/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c", line=line@entry=152, func=func@entry=0x7045ad68 <__func__.15959> "atsam_i2c_set_address_regs", failedexpr=failedexpr@entry=0x7045ae38 "(address & SEVEN_BIT_MASK) == 0") at /home/leon/dev/msc/grisp-software/build/..//rtems/c/src/../../cpukit/libcsupport/src/__assert.c:37
#1  0x703269e0 in atsam_i2c_set_address_regs (regs=<optimized out>, address=<optimized out>, ten_bit_addr=<optimized out>, read_transfer=<optimized out>) at /home/leon/dev/msc/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c:152
#2  0x70576a50 in _Linker_set__Sysinit_rtems_libio_post_driver ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

(gdb) info registers
r0             0x7045adbc       1883614652
r1             0x98     152
r2             0x7045ad68       1883614568
r3             0x7045ae38       1883614776
r4             0x709c7360       1889301344
r5             0x40018000       1073840128
r6             0x0      0
r7             0xb1d8   45528
r8             0xf745   63301
r9             0x0      0
r10            0x0      0
r11            0x0      0
r12            0x0      0
sp             0x20401100       0x20401100
lr             0x703269e1       1882352097
pc             0x7032a470       0x7032a470 <__assert_func>
xPSR           0x21000023       553648163

(gdb) info args
file = 0x7045adbc "/home/leon/dev/msc/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c"
line = 152
func = 0x7045ad68 <__func__.15959> "atsam_i2c_set_address_regs"
failedexpr = 0x7045ae38 "(address & SEVEN_BIT_MASK) == 0"

(gdb) list
32        const char *file,
33        int         line,
34        const char *func,
35        const char *failedexpr
36      )
37      {
38        rtems_assert_context assert_context = {
39          .file = file,
40          .line = line,
41          .function = func,
(gdb) list +
42          .failed_expression = failedexpr
43        };
44
45        printk("assertion \"%s\" failed: file \"%s\", line %d%s%s\n",
46          failedexpr,
47          file,
48          line,
49          (func) ? ", function: " : "",
50          (func) ? func : ""
51        );
(gdb) list +
52        rtems_fatal( RTEMS_FATAL_SOURCE_ASSERT, (rtems_fatal_code) &assert_context );
53      }
54      #endif
(gdb)
```

c-mauderer, 2018-06-22:
That stack frame looks quite odd:
```
(gdb) bt
#0  __assert_func (file=file@entry=0x7045adbc "/home/leon/dev/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c", line=line@entry=152, func=func@entry=0x7045ad68 <__func__.15959> "atsam_i2c_set_address_regs", failedexpr=failedexpr@entry=0x7045ae38 "(address & SEVEN_BIT_MASK) == 0") at /home/leon/dev/grisp-software/build/..//rtems/c/src/../../cpukit/libcsupport/src/__assert.c:37
#1  0x703269e0 in atsam_i2c_set_address_regs (regs=<optimized out>, address=<optimized out>, ten_bit_addr=<optimized out>, read_transfer=<optimized out>) at /home/leon/dev/grisp-software/build/..//rtems/c/src/lib/libbsp/arm/atsam/i2c/atsam_i2c_bus.c:152
#2  0x70576a50 in _Linker_set__Sysinit_rtems_libio_post_driver ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
- Where is your call to the i2c function?
- Maybe there is a problem with the stack size?
- Another possibility could be that the position isn't resolved properly due to some optimization. Could you try it with a RTEMS BSP that is translated with -O1 or -Og?

For the optimization you can set the environment variable `CFLAGS_OPTIMIZE_V="-Og -g -ffunction-sections -fdata-sections"` during the BSP build (works at least on this BSP).

c-mauderer, 2018-06-22:
Just tested it: It should work with the following commands:
```
> export CFLAGS_OPTIMIZE_V="-Og -g -ffunction-sections -fdata-sections" 
> ./build/build.sh --no-toolchain --no-bootstrap --no-submodules
```
During the BSP build, there should be a lot of lines like
```
arm-rtems5-gcc --pipe -DHAVE_CONFIG_H   -I.. -I/home/christian/rtems/grisp-software/build/b-atsamv/arm-rtems5/c/atsamv/include -I/home/christian/rtems/grisp-software/rtems/cpukit/include -I/home/christian/rtems/grisp-software/rtems/cpukit/score/cpu/arm/include -I/home/christian/rtems/grisp-software/build/..//rtems/c/src/../../cpukit/libfs/src/jffs2/include -D__ECOS '-DKBUILD_MODNAME="JFFS2"'  -Wno-pointer-sign -mthumb -mcpu=cortex-m7 -mfpu=fpv5-d16 -mfloat-abi=hard -Og -g -ffunction-sections -fdata-sections -MT src/jffs2/src/libjffs2_a-gc.o -MD -MP -MF src/jffs2/src/.deps/libjffs2_a-gc.Tpo -c -o src/jffs2/src/libjffs2_a-gc.o `test -f 'src/jffs2/src/gc.c' || echo '/home/christian/rtems/grisp-software/build/..//rtems/c/src/../../cpukit/libfs/'`src/jffs2/src/gc.c
mv -f src/jffs2/src/.deps/libjffs2_a-gc.Tpo src/jffs2/src/.deps/libjffs2_a-gc.Po
```
that now contain a `-Og` instead of `-O2`.
You most likely have to re-build the Erlang VM too. @peerst can say more to that.

lwehmeier, 2018-06-22:
I couldn't get it to rebuild an existing system, when using a fresh copy of `grisp-software` your environment variable works, though.
* rebuilding rtems without optimisations
```
> export CFLAGS_OPTIMIZE_V="-Og -g -ffunction-sections -fdata-sections" 
> ./build/build.sh --no-toolchain --no-bootstrap --no-submodules
```
* rebuilding erlang/otp now fails because the grisp build command expects rtems4.12.  For 20.2 : `checking for arm-rtems4.12-gcc... /home/leon/dev/grisp-software-Og/rtems-install/rtems/5/bin/arm-rtems4.12-gcc` fails, for 21.0: `File grisp/grisp_base/xcomp/erl-xcomp-21.0.conf not found`
```
>rm -rf _grisp _build; rebar3 grisp build && rebar3 compile && rebar3 grisp deploy -n helloworld -v 0.1.0
>cat rebar.config
[...]
{grisp, [
    {otp, [{version, "20.2"}]},
    {build, [
        {toolchain, [
            {directory,"/home/leon/dev/grisp-software-Og/rtems-install/rtems/5/"}
        ]}
    ]},
    {deploy, [
        {destination, "/home/leon/dev/msc/deploy"}
    ]}
]}.
[...]
```

c-mauderer, 2018-06-23:
I wasn't aware that the Erlang port for GRiSP still uses the RTEMS 4.12 Version from some time back. The option to build the BSP in a debug configuration via an environment variable is newer than the 4.12 Version used in the original GRiSP release.

OK: Two options:
1. Maybe @peerst knows whether Erlang works with RTEMS5 already. It's his domain.
2. I can tell you a guide how to build this BSP in RTEMS 4.12 with other optimization options:
  - Check out the correct version (tag 1.0.0?).
  - Then make sure your submodules are up to date by doing:
```
git submodule init
git submodule update
cd rtems-libbsd
git submodule init
git submodule update rtems_waf
cd ..
```
  - After that change the `CFLAGS_OPTIMIZE_V = -O2 -g` line in `rtems/c/src/lib/libbsp/arm/atsam/make/custom/atsamv.cfg` to the optimization you want.
  - Build the tools with `./build/build.sh --no-submodules`

lwehmeier, 2018-06-25:
Update: After a few issues during the upgrade to erlang21.0/rtems5 the error changed a little. On `Og` I no longer get an assertion error but a `fatal extension: source=9` error.
```
fatal extension: source=9, is_internal=0, error=541069432
...
R5   = 0x70848a00 SP  = 0x00000201
R6   = 0x40018000 LR  = 0x70331017
R7   = 0x00000000 PC  = 0x70331018
```

Stacktrace:
```
_User_extensions_Iterate
_User_extensions_Fatal_visitor
fatal_extension
```

`0x70331018` (`ldrb r1 [r3]`) is part of the call to `TWI_WriteByte` in `atsam_i2c_setup_write_transfer`, atsam_i2c_bus.c:188.
Setting a breakpoint to that address: The crash occurs during the 5th call to atsam_i2c_setup_write_transfer when ldrb attempts to load from [r3], r3=0x00000002.
Stacktrace:
```
atsam_i2c_interrupt
atsam_i2c_setup_transfer
atsam_i2c_setup_write_transfer
```
Maybe a pointer/buffer handling issue somewhere?

c-mauderer, 2018-06-26:
> Maybe a pointer/buffer handling issue somewhere?

That would be an explanation. Loading from that address is not impossible on this target but it's unlikely. There is a tightly coupled program memory at 0. I think for GRiSP it has a size of zero so there shouldn't be anything there.

So either it's a bug in the I2C driver or some part of the software overwrites the stack (which is another typical bug that leads to a lot of odd errors). I would like to rule out the second one before taking a too detailed look into the driver. Stack bugs can have a lot of forms and most of the time it's quite hard to track them down if you only see the effects.

I'm not exactly sure where the stack sizes for the Erlang Runtime are set. But I would expect that it is mainly the following line:
https://github.com/grisp/grisp/blob/master/grisp/grisp_base/sys/erl_main.c#L392
Could you increase that to for example 800k and try again? Maybe also increase the minimum stack size here:
https://github.com/grisp/grisp/blob/master/grisp/grisp_base/sys/erl_main.c#L395
@peerst: Did I miss some task creation?

If that doesn't change anything (meaning the error is exactly the same), the driver bug becomes more likely. In that case: I had a short look at the `atsam_i2c_setup_write_transfer`. The only indirect access that caught my attention is the [`*bus->current_msg_byte`](https://github.com/grisp/rtems/blob/8b5a801e85e273aa9778bbf5b41269ae15fceb30/bsps/arm/atsam/i2c/atsam_i2c_bus.c#L188).

Could you print the locals of `atsam_i2c_setup_write_transfer` and `atsam_i2c_setup_transfer` in the error case. And maybe also the `*bus` structure in `atsam_i2c_setup_write_transfer`. That should be something like the following gdb commands (note that I wrote them down without trying them):
```
p/x *bus
info locals
up
info locals
```

lwehmeier, 2018-07-02:
new stacksizes (in `_build/default/lib/grisp/grisp/grisp_base/sys/erl_main.c`): `1024K` and `512K`.
 However, as the grisp runtime is automagically build against the provided RTEMS installation (i.e. I don't know where to configure the rebar3_grisp plugin to use a custom grisp repository for the build), I'm not sure if the stack sizes have actually changed or if the grisp build process used a different copy of erl_main.c.

Result: Still crashes
```
fatal extension: source=9, is_internal=0, error=541069432

R0   = 0x40018000 R8  = 0x00000001
R1   = 0x00010000 R9  = 0x00000000
R2   = 0x00000000 R10 = 0x00000000
R3   = 0x00000002 R11 = 0x00000000
R4   = 0x40018000 R12 = 0x705e3f18
R5   = 0x71118a00 SP  = 0x00000201
R6   = 0x40018000 LR  = 0x70331017
R7   = 0x00000000 PC  = 0x70331018
XPSR = 0x410f0023 VEC = 0x00000004
```
locals and bus:
```
```

peerst, 2018-07-02:
The easiest way to have a custom `grisp` repo is to have a clone in the `_checkouts` directory.

See https://www.rebar3.org/docs/dependencies#section-checkout-dependencies

peerst, 2018-07-02:
But in general I don't think stackoverflow is the issue:  we have RTEMS stack checker on and have not even been close so far.

c-mauderer, 2018-07-02:
@peerst: The stack checker can catch some stack errors but doesn't catch all of them. But changing the stack sizes of the Init task should have already moved some memory layout. It seems to be still the same error so at least a wild pointer is slightly less likely. If you are sure that there is plenty of stack left, the next steps would be to have a more thorough look at the I2C driver.

I don't really know that special I2C driver well. But from a first glance, the `bus->msg_todo = -1` (or `0xffffffff`) looks a little odd. The only time I found where msg_todo is decremented is here: https://github.com/grisp/rtems/blob/8b5a801e85e273aa9778bbf5b41269ae15fceb30/bsps/arm/atsam/i2c/atsam_i2c_bus.c#L110
But in the normal case, the program flow should have hit the following `bus->msg_todo == 0` and therefore the interrupts should be disabled for the I2C: https://github.com/grisp/rtems/blob/8b5a801e85e273aa9778bbf5b41269ae15fceb30/bsps/arm/atsam/i2c/atsam_i2c_bus.c#L269

Two ideas:

- Either some odd timing problem where one interrupt is already pending? It's unlucky that the irqstatus has been optimized out. Maybe you could add a `assert(regs->TWIHS_IDR != 0xFFFFFFFF);` in the else case here: https://github.com/grisp/rtems/blob/8b5a801e85e273aa9778bbf5b41269ae15fceb30/bsps/arm/atsam/i2c/atsam_i2c_bus.c#L276. If that assert triggers, the interrupt shouldn't have happened in the first place and it's definitively a problem with the interrupt logic in the driver.

- Or the message count has been wrong in the first place. Maybe you could check the value of it when entering this function: https://github.com/grisp/rtems/blob/8b5a801e85e273aa9778bbf5b41269ae15fceb30/bsps/arm/atsam/i2c/atsam_i2c_bus.c#L282

eproxus, 2018-07-03:
@lwehmeier @peerst You really want to override the `erl_main.c` in your own `grisp/grisp_base/sys/` folder in your own project. This will get picked up by the Rebar 3 plug-in when building Erlang.

lwehmeier, 2018-07-17:
Sorry, the replies got lost in my GitHub feed..
@eproxus That makes it quite a bit easier, thanks. Didn't change the results, though.
@c-mauderer The number of messages passed to the transfer function is correct, as is the msgs pointer itself (can be dereferenced and contains the expected messages).
 `assert(regs->TWIHS_IDR != 0xFFFFFFFF);` right before the setup_transfer call did not trigger.

I suspect there that there are some issues on buffer handling somewhere (no idea whether in grisp or rtems), as I've been having other potentially related I2C issues. E.g., when a read or write request fails (NAK), no error is reported but the grisp_i2c:msgs function returns a "valid" value (232 mostly). After that it takes a few (sometimes one, sometimes up to 5) dummy writes before the TWI module sends out correct data again. I assume that the transmit hardware registers (and potentially buffers?) are not cleared if an error occurs. The same thing also happens occasionally when a TWI slave extends the SCL low period on fast bus transfers

peerst, 2018-07-19:
Is there some reasonable way to reproduce this. I’d like to debug this a bit

lwehmeier, 2018-07-20:
The easiest way to reproduce this would probably be to create an "empty" helloworld project, deploy that, connect to the grisp board and run the commands I posted earlier. The configuration for my test project (generated following the helloworld tutorial):
```
>> cat rebar.config 
{deps, [
    grisp,
    {epmd, {git, "https://github.com/erlang/epmd", {ref, "4d1a59"}}}
  ]}.

{erl_opts, [debug_info]}.

{plugins, [
  rebar3_grisp
]}.

{grisp, [
    {otp, [{version, "21.0"}]},
    {deploy, [
        {destination, "/home/leon/grisp/deploy"}
    ]}
]}.

{shell, [{apps, []}]}.

{relx, [
    {release, {helloworld, "0.1.0"}, [{epmd, none},  helloworld]}
]}.
```

For testing, my board is connected to an ina219 at 0x40, but that shouldn't be needed.
To crash it (might need to be run several times since I upgraded to 21.0):
```
>>grisp_i2c:msgs([16#40, {write, <<16#42>>}, {write, <<16#23, 16#24, 16#25>>}]).
```

Additionally, the use of grisp_i2c after failed reads (read request/multibyte read request, NAK response) can sometimes (i.e. not reproducibly) cause weird behaviour (I2C master stuck at low, even after periphery is disconnected (No idea how that's possible tbh)) or crashes similar to the ones reported here.

[image]
This happened when testing an i2c driver for the tca9548 earlier today. The board crashed (assertion failed) and SCL and SDA were stuck at low. (ignore the undershooting, the scope just had a bad gnd connection)

lwehmeier, 2018-07-20:
I created a [repo with some instructions](https://github.com/lwehmeier/grisp_i2c_crash) for a more complex example that crashes my board nearly every time.

sebhub, 2018-09-28:
Do you use I2C transfers with a finite timeout value? It is controlled by the I2C_TIMEOUT IO control.

sylane, 2024-07-01:
@lwehmeier Is this still an issue ?