grisp/grisp #52: Make a way to find out the Wifi MAC address

peerst, 2019-05-10:


Laymer, 2019-06-06:
Hello 🙂 

Would be happy to help on this issue ! I have also tested that `inet` provides the correct `hwaddr` on Mac, and the field is missing from the result when called on GRiSP. 

I came across something in the OTP code that might be related to this : 

https://github.com/grisp/otp/blob/fe2b1323a3866ed0a9712e9d12e1f8f84793ec47/erts/emulator/drivers/common/inet_drv.c#L5327

Originally found in this StackOverflow post : 

https://stackoverflow.com/questions/9292967/erlang-otp-fails-to-compile-on-solaris-11-11-11

I have not tried to modify this in any way yet but the part that says `fix include autoconf later` just caught my attention and I don't know if it could affect this issue somehow?

peerst, 2019-06-06:
You found the correct position in the code, thank you thats already very useful.

So now the next questions are:

Does RTEMS have a field `ifr_hwaddr` in the struct

    if yes ->
          
    why is it not detected by autoconf?

    if no ->

    can we add it to RTEMS?

Since the network stack is from FreeBSD I would assume the field to be there so it might be a autoconf detection that doesn't work for RTEMS

peerst, 2019-06-13:
We got the information from @sebhub that as we assumed it is supposed to work on RTEMS/libbsd exactly as on FreeBSD.  So I built OTP on FreeBSD 12 and checked if the hwaddr is shown there.

```
1> inet:getifaddrs().
{ok,[{"vtnet0",
      [{flags,[up,broadcast,running,multicast]},
       {addr,{157,230,119,28}},
       {netmask,{255,255,240,0}},
       {broadaddr,{157,230,127,255}},
       {addr,{65152,0,0,0,4333,37887,65237,54044}},
       {netmask,{65535,65535,65535,65535,0,0,0,0}},
       {addr,{10,19,0,14}},
       {netmask,{255,255,0,0}},
       {broadaddr,{10,19,255,255}},
       {hwaddr,[18,237,147,213,211,28]}]},
     {"lo0",
      [{flags,[up,loopback,running,multicast]},
       {addr,{0,0,0,0,0,0,0,1}},
       {netmask,{65535,65535,65535,65535,65535,65535,65535,65535}},
       {addr,{65152,0,0,0,0,0,0,1}},
       {netmask,{65535,65535,65535,65535,0,0,0,0}},
       {addr,{127,0,0,1}},
       {netmask,{255,0,0,0}}]}]}
2> 
```

So we know it works on FreeBSD.  But Erlang uses a different mechanism to get at the hwaddr since:

```
/* Define to 1 if `ifr_enaddr' is a member of `struct ifreq'. */
/* #undef HAVE_STRUCT_IFREQ_IFR_ENADDR */

/* Define to 1 if `ifr_hwaddr' is a member of `struct ifreq'. */
/* #undef HAVE_STRUCT_IFREQ_IFR_HWADDR */

```

peerst, 2019-06-13:
There are 4 mechanisms OTP uses to discover the hardware address:  

https://github.com/grisp/otp/blob/fe2b1323a3866ed0a9712e9d12e1f8f84793ec47/erts/emulator/drivers/common/inet_drv.c

(Search for INET_IFOPT_HWADDR, that's the key used to write the hwaddr field towards Erlang side)

We figured that two of them are not used on FreeBSD (IFR_HWADDR and IFR_ENADDR).

The remaining two are using DLPI (which is assumedly used on Solaris and descendants) but:

```
/* Define to 1 if you have the `dlpi' library (-ldlpi). */
/* #undef HAVE_LIBDLPI */

/* Define to 1 if you have the <libdlpi.h> header file. */
/* #undef HAVE_LIBDLPI_H */
```

peerst, 2019-06-13:
So that leaves the forth one:

https://github.com/grisp/otp/blob/fe2b1323a3866ed0a9712e9d12e1f8f84793ec47/erts/emulator/drivers/common/inet_drv.c#L5346

config.h:
```
/* Define to 1 if you have the `getifaddrs' function. */
#define HAVE_GETIFADDRS 1
```

and /usr/include/sys/socket.h:

```
/usr/include/sys/socket.h:#define	AF_LINK		18		/* Link layer interface */
```

So now we know for sure that on FreeBSD this mechanism is used

sebhub, 2019-06-13:
getifaddrs() is supported by libbsd.

peerst, 2019-06-13:
So for RTEMS4 based toolchains we already know that

HAVE_GETIFADDRS is not defined in config.h

Need to figure out the failing test case

eproxus, 2019-06-13:
Relevant `configure` failure:

```
...
configure:17517: checking for getifaddrs
configure:17517: /opt/grisp/grisp-software/3cb8992961/rtems-install/rtems/5/bin/arm-rtems5-gcc -pipe -B /opt/grisp/grisp-software/3cb8992961/rtems-install/rtems/5/arm-rtems5/atsamv/lib -specs bsp_specs -qrtems -o conftest -Wall -Wextra -Wconversion -Wformat-security -Wformat=2 -Wshadow -Wcast-qual -Wcast-align -Wredundant-decls -Wstrict-prototypes -Wbad-function-cast -Wno-unused-parameter -Wno-redundant-decls -Wno-sign-conversion -Wno-sign-compare -mthumb -mcpu=cortex-m7 -mfpu=fpv5-d16 -mfloat-abi=hard -g -ffunction-sections -fdata-sections -O0 -I/Users/alind/Stritzinger/Code/GRiSP/example/_grisp/otp/21.0/build/erts/arm-unknown-rtems5  -fno-tree-copyrename   -DUSE_THREADS -D_THREAD_SAFE -D_REENTRANT -DPOSIX_THREADS -mthumb -mcpu=cortex-m7 -mfpu=fpv5-d16 -mfloat-abi=hard -Wl,--gc-sections -qnolinkcmds -T linkcmds.sdram -Wl,-Map -Wl,linker.map conftest.c -lm -lgrisp -linih -lbsd     >&5
conftest.c:214:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 char getifaddrs ();
 ^~~~
conftest.c:223:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 main ()
 ^~~~
/opt/grisp/grisp-software/3cb8992961/rtems-install/rtems/5/arm-rtems5/atsamv/lib/librtemscpu.a(bdbuf.o): In function `rtems_mutex_lock':
/opt/grisp/grisp-software/3cb8992961/rtems/cpukit/include/rtems/thread.h:54: undefined reference to `rtems_bdbuf_configuration'
/opt/grisp/grisp-software/3cb8992961/rtems-install/rtems/5/arm-rtems5/atsamv/lib/librtemscpu.a(bdbuf.o): In function `rtems_bdbuf_set_block_size':
/opt/grisp/grisp-software/3cb8992961/build/..//rtems/c/src/../../cpukit/libblock/src/bdbuf.c:2880: undefined reference to `rtems_bdbuf_configuration'
collect2: error: ld returned 1 exit status
configure:17517: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME ""
| #define PACKAGE_TARNAME ""
| #define PACKAGE_VERSION ""
| #define PACKAGE_STRING ""
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| #define ASSUMED_CACHE_LINE_SIZE 64
| #define ERTS_SAVED_COMPILE_TIME 1
| #define ERTS_ENABLE_MSACC 1
| #define STDC_HEADERS 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_STAT_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_STRING_H 1
| #define HAVE_MEMORY_H 1
| #define HAVE_STRINGS_H 1
| #define HAVE_INTTYPES_H 1
| #define HAVE_STDINT_H 1
| #define HAVE_UNISTD_H 1
| #define SIZEOF_VOID_P 4
| #define HAVE_LIBM 1
| #define ETHR_TRUST_GCC_ATOMIC_BUILTINS_MEMORY_BARRIERS 0
| #define HAVE_PTHREAD_H 1
| #define HAVE_CLOCK_GETRES 1
| #define ETHR_HAVE_CLOCK_GETTIME_MONOTONIC 1
| #define ETHR_MONOTONIC_CLOCK_ID CLOCK_MONOTONIC
| #define ETHR_PTHREADS 1
| #define ETHR_HAVE_PTHREAD_H 1
| #define ETHR_HAVE_SCHED_H 1
| #define ETHR_HAVE_SYS_TIME_H 1
| #define ETHR_TIME_WITH_SYS_TIME 1
| #define ETHR_HAVE_USABLE_PTHREAD_STACK_MIN 1
| #define ETHR_HAVE_PTHREAD_SPIN_LOCK 1
| #define ETHR_HAVE_SCHED_YIELD 1
| #define ETHR_SCHED_YIELD_RET_INT 1
| #define ETHR_HAVE_PTHREAD_ATTR_SETGUARDSIZE 1
| #define ETHR_HAVE_PTHREAD_COND_TIMEDWAIT_MONOTONIC 1
| #define SIZEOF_SHORT 2
| #define SIZEOF_INT 4
| #define SIZEOF_LONG 4
| #define SIZEOF_LONG_LONG 8
| #define SIZEOF___INT128_T 0
| #define ETHR_HAVE___sync_synchronize ~0
| #define ETHR_HAVE___sync_add_and_fetch 4
| #define ETHR_HAVE___sync_fetch_and_and 4
| #define ETHR_HAVE___sync_fetch_and_or 4
| #define ETHR_HAVE___sync_val_compare_and_swap 4
| #define ETHR_HAVE___atomic_store_n 12
| #define ETHR_HAVE___atomic_load_n 12
| #define ETHR_HAVE___atomic_add_fetch 12
[... 154 lines cut ...]
| #undef getifaddrs
| 
| /* Override any GCC internal prototype to avoid an error.
|    Use char because int might match the return type of a GCC
|    builtin and then its argument prototype would still apply.  */
| #ifdef __cplusplus
| extern "C"
| #endif
| char getifaddrs ();
| /* The GNU C library defines this for functions which it implements
|     to always fail with ENOSYS.  Some functions are actually named
|     something starting with __ and the normal name is an alias.  */
| #if defined __stub_getifaddrs || defined __stub___getifaddrs
| choke me
| #endif
| 
| int
| main ()
| {
| return getifaddrs ();
|   ;
|   return 0;
| }
configure:17517: result: no
```

peerst, 2019-06-13:
@sebhub so it fails because of some bdbuf stuff missing in the configure test case and not because `getifaddrs()` is missing.  Any idea what could have happened here?

peerst, 2019-06-13:
Apparently one cannot make a small program using getifaddrs without having a bdbuf configuration

sebhub, 2019-06-13:
> @sebhub so it fails because of some bdbuf stuff missing in the configure test case and not because `getifaddrs()` is missing. Any idea what could have happened here?

Try to add "-lrtemsbsp -lrtemscpu -lrtemsdefaultconfig" to the linker flags. You already have "-lbsd"?

eproxus, 2019-06-13:
@sebhub Trying those flags I get the following error during configure:
```
configure:2598: checking whether the C compiler works
configure:2620: /opt/grisp/grisp-software/3cb8992961/rtems-install/rtems/5/bin/arm-rtems5-gcc -pipe -B /opt/grisp/grisp-software/3cb8992961/rtems-install/rtems/5/arm-rtems5/atsamv/lib -specs bsp_specs -qrtems -Wall -Wextra -Wconversion -Wformat-security -Wformat=2 -Wshadow -Wcast-qual -Wcast-align -Wredundant-decls -Wstrict-prototypes -Wbad-function-cast -Wno-unused-parameter -Wno-redundant-decls -Wno-sign-conversion -Wno-sign-compare -mthumb -mcpu=cortex-m7 -mfpu=fpv5-d16 -mfloat-abi=hard -g -ffunction-sections -fdata-sections -O0  -mthumb -mcpu=cortex-m7 -mfpu=fpv5-d16 -mfloat-abi=hard -Wl,--gc-sections -qnolinkcmds -T linkcmds.sdram -Wl,-Map -Wl,linker.map conftest.c -lgrisp -linih -lbsd -lrtemsbsp -lrtemscpu -lrtemsdefaultconfig >&5
conftest.c:11:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 main ()
 ^~~~
/opt/grisp/grisp-software/3cb8992961/rtems-install/rtems/5/lib/gcc/arm-rtems5/7.3.0/../../../../arm-rtems5/bin/ld: cannot find -lrtemsdefaultconfig
collect2: error: ld returned 1 exit status
```

sebhub, 2019-06-14:
It seems your RTEMS version is not new enough to provide a -lrtemsdefaultconfig.

Who calls the main() function in your Autoconf tests?

eproxus, 2019-06-14:
@sebhub `conftest.c` looks like this:
```c
/* confdefs.h */
#define PACKAGE_NAME ""
#define PACKAGE_TARNAME ""
#define PACKAGE_VERSION ""
#define PACKAGE_STRING ""
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""
/* end confdefs.h.  */

int
main ()
{

  ;
  return 0;
}
```

sebhub, 2019-06-14:
Yes, and who calls this main function?

eproxus, 2019-06-14:
The linking fails. No-one is calling the function.

peerst, 2019-06-14:
These are normal autoconf tests, so no-one calls main().  The tests like most of those generate little programs with a main() and see if it links.  Since normally main() is called implicitly at program startup

sebhub, 2019-06-17:
I would provide a stub code so that all link-time tests by the configure script pass or fail as you want. Alternatively, you could copy a hard coded configuration header into the build tree.

peerst, 2019-06-17:
Well we would really like the tests to behave correctly.  Fail for stuff that isn't there and succeed for things that is.  Because 99% of the time it works just fine just this one is broken.  And there are really many of these tests in Erlang/OTP

What was this about your earlier suggestion to pass `-lrtemsbsp -lrtemscpu -lrtemsdefaultconfig` on the linker command line? Why don't we have some of those?  Would these solve the problem, without breaking the actual linking?

peerst, 2019-06-17:
@sebhub Ah missed the comment about RTEMS version being too old

peerst, 2019-06-17:
So I'd say since this is no blocking point at the moment we try to upgrade RTEMS version when we have the resources and then use -lrtemsdefaultconfig etc. to fix this

sebhub, 2019-06-18:
I had a look into the default configuration and using -lrtemsdefaultconfig will likely not fix this issue.

You need a configuration which enables a use of getifaddrs() called by a main() function. The first step is to figure out who calls main() in your setup. Could you give me a linker map file of the configuration test?

eproxus, 2019-06-26:
@sebhub Not sure what you mean. In the minimal example above, no-one calls `main()` since it's just a compiler/linker test. In our real setup, I guess the bootloader calls `main()` when our RTEMS executable is started?

sebhub, 2019-06-26:
I need the linker map file.

Someone in your setup must provide a RTEMS application configuration with a initialization task which calls main().

peerst, 2019-06-26:
For the normal linking we have a explicit Init task which starts up Erlang.  But this is not picked up by the configure tests and it can’t because it’s not compiled yet at this point and would require the whole Erlang VM.  Kind of a hen and egg problem

sebhub, 2019-06-27:
RTEMS provides a default configuration which calls main(), however, this default configuration doesn't include the libblock. I guess the default configuration is used. The linker map file would show this. In recent RTEMS versions, the default configuration is available in a separate library (librtemsdefaultconfig.a) and not in librtemscpu.a.

sylane, 2024-07-01:
@peerst Is this still relevant ?

peerst, 2024-07-02:
Well it segwayed into other things.  Do we have hwaddr now?  Then it would be resolved, otherwise I don't see why it wouldn't still be useful

Laymer, 2024-07-02:
I have not specifically tried to read the MAC address on GRiSP 2 but I assumed that had to be possible for DHCP to work out of the box.

Maybe I am missing something but I am quite sure the hwaddr is available in the new version 👍

peerst, 2024-07-02:
DHCP is implemented in C and starts running before the Erlang VM.  This is about hwaddr being (correctly) in the return value of the mentioned Erlang level call

sylane, 2024-07-03:
It seems to be working now:

sylane, 2024-07-03:
It seems to be fixed in current version.

closed 2024-07-03 by sylane

Laymer, 2024-07-04:
> _DHCP is implemented in C and starts running before the Erlang VM.  This is about hwaddr being (correctly) in the return value of the mentioned Erlang level call_

Thanks for clarifying @peerst  ☺️ I mixed up the levels 😅