grisp/grisp #63: Create a way to have the hostname be auto-generated

peerst, 2019-09-06:
### Description

Starting the same release on multiple GRiSP boards connected to the same
Wifi should be more easy.  We should use a autogenerated hostname derived
from the serial number (which is somewhere in the EEPROM if I remember correctly).

e.g.  board with serial number 123 might get hostname grisp0123

Also make this work with GRiSP 2 boards (maybe we just continue serial numbers there)

eproxus, 2019-09-10:
There is a workaround for setting the hostname as an argument to `deploy`:

It's possible to use environment variables in the templates. So one could do the following:

```
[network]
{{#env.GRISP_HOSTNAME}}
hostname={{env.GRISP_HOSTNAME}}
{{/env.GRISP_HOSTNAME}}
```

And then run deploy:
```sh
$ GRiSP_HOSTNAME=foobar rebar3 grisp deploy ...
```

Laymer, 2019-09-15:
Hello :) 

I might be able to provide some help for this, as I have been helped by @pedroAkos for the Yggdrasil integration that derives an IP based on the `hwaddr` of a GRiSP board. Currently I have the following output at boot time : 

```
Interface MAC address is 38:1d:d9:46:2d:e6
INFO: CONNECTING TO INTERFACE -> 'wlan0'
Hostname -> '169-254-215-181'
Interface: wlan0        Address: 169.254.215.181
Ip address -> '169.254.215.181'
starting erlang runtime
Eshell V10.0  (abort with ^G)
(achlys@169-254-215-181)1> node().
'achlys@169-254-215-181'
(achlys@169-254-215-181)2> 1988-01-01T00:10:41.993428+00:00 critical: Formation result : {ok,['achlys@169-254-215-181']}
(achlys@169-254-215-181)2> inet:getifaddrs().
{ok,[{"lo0",
      [{flags,[up,loopback,running,multicast]},
       {addr,{127,0,0,1}},
       {netmask,{255,255,255,0}}]},
     {"wlan0",
      [{flags,[up,broadcast,running,multicast]},
       {addr,{169,254,215,181}},
       {netmask,{255,255,0,0}},
       {broadaddr,{169,254,255,255}}]}]}
```

In the output above the hostname is `"169-254-215-181"` but it could be certainly changed to something else (this was only for debugging). But I have just tested it and I am able to connect my laptop to that board and send/receive messages using `gen_udp`. So if this is something that might be useful feel free to let me know 🙂

PS :  thanks @eproxus for the deployment trick !

closed 2024-07-01 by sylane