grisp/grisp #73: [QUESTION] Wiki & embedded mode in grisp.ini

Laymer, 2020-03-20:
### Versions

* OS: macOS 10.15.4
* GRiSP Toolchain: 
* Erlang: 22.0
* Rebar: 3.13.1

### Description

Hello 🙂 

I have recently noticed that the Wiki page describing how to connect to a GRiSP board over Wi-Fi does not set the `-mode embedded` flag in the 
[grisp.ini](https://github.com/grisp/grisp/wiki/Connecting-over-WiFI#grisp-ini) section : 

```
[erlang]
args = erl.rtems -- -home . -pa . -root {{release_name}} -boot {{release_name}}/releases/{{release_version}}/{{release_name}} -internal_epmd epmd_sup -kernel inetrc "./erl_inetrc" -sname {{release_name}} -setcookie MyCookie
```

And I've seen that it has been [set as default](https://github.com/grisp/grisp/commit/28de1dd1198389442791e8c90fd28ab5e529994d) in the `grisp/grisp_base/files/grisp.ini.mustache` file. Reading about the [`code:get_mode/0`](http://erlang.org/doc/man/code.html#get_mode-0) function, I saw that embedded mode has implications for code loading & releases handling.

So just wondering if there is a specific reason that I am missing why the mode must be set to interactive when connecting over Wi-Fi, or is it something unrelated/typo in the Wiki ? 🙂 

Thanks in advance,

Igor

peerst, 2020-03-21:
There is no specific reason, both modes work in any case.  But we should be more explicit and/or consistent

Laymer, 2020-03-22:
Thank you ! :) 
I konw there's no way to PR for the Wiki section, so can I help in some way or maybe close the issue? 🙂

eproxus, 2020-03-23:
I've fixed the wiki, using the same format as the `grisp.ini.mustache` template.

closed 2020-03-23 by eproxus