grisp/grisp #99: [GRiSP1 support] Attempt to deploy robot example on grisp_base

dagyu, 2022-03-02:
### Versions

* OS: Ubuntu 20.04.4 LTS
* Rebar: rebar 3.18.0 on Erlang/OTP 23 Erts 11.2.2.3
* rebar3_grisp (2.2.2)

### Problem statement

I'm trying to deploy robot example for `grisp_base` so I follow these two possible ways:
- Deploy my code for `grisp_base` adding this line `{platform, grisp_base}` to my `rebar.config`
- Deploy my code for `grisp2` and try if it works in `grisp_base`

but nothing seems to work for me.

#### 1) Deploy my code for `grisp_base`
 I just add `{platform, grisp_base}` on my `rebar.config` and then when I type 
`rebar3 grisp deploy` the plugin is not able to find a pre-built package for my OTP:

```
 rebar3 grisp deploy                                                                                                                      1 ↵
===> Verifying dependencies...
===> Analyzing applications...
===> Compiling robot
* Resolving OTP version
    23.3.4.4 (requirement was "23")
* Downloading pre-built OTP package (3593d1d1)
===> Could not find a pre-built package for Erlang 23.3.4.4!
Possible causes:
  - You have custom C sources and need to do a manual build:
      rebar3 grisp build
  - There is no pre-built package matching this hash:
      3593d1d1bc7d6fbb265c295163ad2a105842673b4402161a1f8eff061a5dd5fa
```

I chose OTP version 23.3.4.4 because when is in the list of pre-build OTP versions for platform 'grisp_base'. I fetch this list typing `rebar3 grisp package list -p grisp_base`, and the output is:

```
===> GRiSP pre-built OTP versions for platform 'grisp_base'
Version
20.2
21.0
21.0-rc1
22.0
23.3.4.4
```

I can't use the older versions because the last version of `rebar3_grisp` not compile with them.  

Versions:
* Erlang: OTP 23.3.4.4

#### 2) Deploy my code for `grisp2` 

When I try to boot my `grisp.ini` file deployed for `grisp2` the boot fail. This is my output:
```
[/media/mmcsd-0-0] # boot grisp.ini
boot grisp.ini
boot: open file "grisp.ini"... done
boot: read file "grisp.ini"... received 199 bytes

----------------
GRISP bootloader
----------------
Version: bootloader-V1.0.2 (Build Mi 6. Dez 15:09:49 UTC 2017)
nexus0: <RTEMS Nexus device>
at91_mci0: <MCI mmc/sd host bridge> on nexus0
mmc0: <MMC/SD bus> on at91_mci0
media listener: event = DISK ATTACH, state = INQUIRY, src = mmcsd
media listener: event = DISK ATTACH, state = SUCCESS, src = mmcsd, dest = /dev/mmcsd-0
mmedia listener: event = MOUNT, state = INQUIRY, src = /dev/mmcsd-0
mcsd0: 8GB <SDHC USDU1 2.0 SN 43159Emedia listener: event = MOUNT, state = FAILED, src = /dev/mmcsd-0
5media listener: event = PARTITION INQUIRY, state = INQUIRY, src = /dev/mmcsd-0
8media listener: event = PARTITION INQUIRY, state = SUCCESS, src = /dev/mmcsd-0
 media listener: event = PARTITION ATTACH, state = INQUIRY, src = /dev/mmcsd-0
Mmedia listener: event = PARTITION ATTACH, state = SUCCESS, src = /dev/mmcsd-0, dest = /dev/mmcsd-0-0
Fmedia listener: event = MOUNT, state = INQUIRY, src = /dev/mmcsd-0-0
G 05/2018 by 116 J`> at mmc0 20.5MHz/4bit/1-block
media listener: event = MOUNT, state = SUCCESS, src = /dev/mmcsd-0-0, dest = /media/mmcsd-0-0
[zone: udpcb] kern.ipc.maxsockets limit reached
waiting for SD...

boot: press key to enter service mode
boot: press key to enter service mode
boot: press key to enter service mode

RTEMS Shell on /dev/console. Use 'help' to list commands.
```

Versions:
* GRiSP Toolchain: grisp_otp_build_23.3.4.11_a80bccc41bebe9f21b8623d29b0c801277928295ab1f81531e37b2eb5353521e.tar.gz
* Erlang: OTP 23.3.4.11

### Final question

Someone could help me?

eproxus, 2022-03-02:
@dagyu I've uploaded a new pre-built package for OTP 23.3.4.11 for `grisp_base`. You can see it here:
```console
$ rebar3 grisp package list --platform=grisp_base --columns=version,hash
===> GRiSP pre-built OTP versions for platform 'grisp_base'
Version    Hash
...
23.3.4.11  3593d1d1bc7d6fbb265c295163ad2a105842673b4402161a1f8eff061a5dd5fa
...
```

eproxus, 2022-03-02:
Try and see if that works for you 🙂

dagyu, 2022-03-02:
> Try and see if that works for you slightly_smiling_face

Thank you now it works!

closed 2022-03-02 by eproxus