grisp/grisp #100: Boot failing when try to connect over WiFI

dagyu, 2022-03-03:
### Versions

* OS: Ubuntu 20.04.4 LTS
* GRiSP Toolchain: a80bccc41bebe9f21b8623d29b0c801277928295ab1f81531e37b2eb5353521e
* Erlang: 23.3.4.11
* Rebar: 3.18.0 on Erlang/OTP 23 Erts 11.2.2.10
* Board: grisp_base

### Description

I'm trying to connect my grisp_base board over WiFI but I have some problems. I followed the wiki but when my bootloader start it fails with this output message:

```
boot: open file "/media/mmcsd-0-0/grisp.bin"... failed
```

In fact the file `/media/mmcsd-0-0/grisp.bin` doesn't exist on my sd card. 

Just to add further infos I tried to add `erl_inetrc`, `grisp.ini.mustache` and `wpa_supplicant.conf` on these paths:
1. grisp/default/common/deploy/files/ [WIKI]
2. grisp/grisp_base/common/deploy/files/ [WIKI]
3. grisp/grisp_base/files/ [TWICHT VIDEO https://www.twitch.tv/videos/419911636]

But only the first one works that means that copy `erl_inetrc` and `wpa_supplican.conf` inside my deploy, but it still doesn't seem to add the right grisp.ini. I suppose that the right `grisp.ini` should be derived from the template `grisp.ini.mustache` but actually that doesn't happen.

eproxus, 2022-03-09:
The instructions are unfortunately outdated. (1) is overridden by (2), and (3) is outdated and no longer used. Therefore (2) should be the way to go.

It should work if you put all files in `grisp/grisp_base/common/deploy/files` since that is where the original file comes from: https://github.com/grisp/grisp/blob/master/grisp/grisp_base/common/deploy/files/grisp.ini.mustache

Can you try that?

dagyu, 2022-03-09:
I tried but I have yet some problems. In particular when I the boot start fails with this output:

```
----------------
GRISP bootloader
----------------
Version: bootloader-V1.0.2 (Build Mi 6. Dez 15:09:49 UTC 2017)
nexus0: <RTEMS Nexus device>
at91_mci0: <MCI mmc/sd host bridge> on nexus0
mmc0: <MMC/SD bus> on at91_mci0
media listener: event = DISK ATTACH, state = INQUIRY, src = mmcsd
media listener: event = DISK ATTACH, state = SUCCESS, src = mmcsd, dest = /dev/mmcsd-0
mmedia listener: event = MOUNT, state = INQUIRY, src = /dev/mmcsd-0
mcsd0: 8GB <SDHC USDU1 2.0 SN 43159E58 MFG 05/2018 by 116 J`> at mmc0 20.5MHz/4media listener: event = MOUNT, state = FAILED, src = /dev/mmcsd-0
bmedia listener: event = PARTITION INQUIRY, state = INQUIRY, src = /dev/mmcsd-0
imedia listener: event = PARTITION INQUIRY, state = SUCCESS, src = /dev/mmcsd-0
tmedia listener: event = PARTITION ATTACH, state = INQUIRY, src = /dev/mmcsd-0
/media listener: event = PARTITION ATTACH, state = SUCCESS, src = /dev/mmcsd-0, dest = /dev/mmcsd-0-0
1media listener: event = MOUNT, state = INQUIRY, src = /dev/mmcsd-0-0
-block
media listener: event = MOUNT, state = SUCCESS, src = /dev/mmcsd-0-0, dest = /media/mmcsd-0-0
[zone: udpcb] kern.ipc.maxsockets limit reached
waiting for SD...

boot: press key to enter service mode
boot: press key to enter service mode
boot: press key to enter service mode
boot: open file "/media/mmcsd-0-0/rosie_grisp/erts-11.2.2.10/bin/beam.bin"... failed
```

In effect the file `/media/mmcsd-0-0/rosie_grisp/erts-11.2.2.10/bin/beam.bin` doesn't exist on my build, this the `ls` command of the folder in my sd

```
[/media/mmcsd-0-0] # ls rosie_grisp/erts-11.2.2.10/bin
beam                erl                 start
beam.smp            erl.src             start.src
beam.zImage         imx6ul-grisp2.dtb   start_erl.src
```

Tell me if you need some further infos or logs.
```

dagyu, 2022-03-09:
Ok maybe I understand the problem, the image path of the `grisp.ini.mustache` that you linked is different. I'll try with the right one. So I think that also `grisp.ini.mustache` on the WIKI is outdated.

> I tried but I have yet some problems. In particular when I the boot start fails with this output:
> 
> ```
> ----------------
> GRISP bootloader
> ----------------
> Version: bootloader-V1.0.2 (Build Mi 6. Dez 15:09:49 UTC 2017)
> nexus0: <RTEMS Nexus device>
> at91_mci0: <MCI mmc/sd host bridge> on nexus0
> mmc0: <MMC/SD bus> on at91_mci0
> media listener: event = DISK ATTACH, state = INQUIRY, src = mmcsd
> media listener: event = DISK ATTACH, state = SUCCESS, src = mmcsd, dest = /dev/mmcsd-0
> mmedia listener: event = MOUNT, state = INQUIRY, src = /dev/mmcsd-0
> mcsd0: 8GB <SDHC USDU1 2.0 SN 43159E58 MFG 05/2018 by 116 J`> at mmc0 20.5MHz/4media listener: event = MOUNT, state = FAILED, src = /dev/mmcsd-0
> bmedia listener: event = PARTITION INQUIRY, state = INQUIRY, src = /dev/mmcsd-0
> imedia listener: event = PARTITION INQUIRY, state = SUCCESS, src = /dev/mmcsd-0
> tmedia listener: event = PARTITION ATTACH, state = INQUIRY, src = /dev/mmcsd-0
> /media listener: event = PARTITION ATTACH, state = SUCCESS, src = /dev/mmcsd-0, dest = /dev/mmcsd-0-0
> 1media listener: event = MOUNT, state = INQUIRY, src = /dev/mmcsd-0-0
> -block
> media listener: event = MOUNT, state = SUCCESS, src = /dev/mmcsd-0-0, dest = /media/mmcsd-0-0
> [zone: udpcb] kern.ipc.maxsockets limit reached
> waiting for SD...
> 
> boot: press key to enter service mode
> boot: press key to enter service mode
> boot: press key to enter service mode
> boot: open file "/media/mmcsd-0-0/rosie_grisp/erts-11.2.2.10/bin/beam.bin"... failed
> ```
> 
> In effect the file `/media/mmcsd-0-0/rosie_grisp/erts-11.2.2.10/bin/beam.bin` doesn't exist on my build, this the `ls` command of the folder in my sd
> 
> ```
> [/media/mmcsd-0-0] # ls rosie_grisp/erts-11.2.2.10/bin
> beam                erl                 start
> beam.smp            erl.src             start.src
> beam.zImage         imx6ul-grisp2.dtb   start_erl.src
> ```
> 
> I don't know it it is useful to figure out the problem but this is my grisp.ini
> 
> ```
> [/media/mmcsd-0-0] # cat grisp.ini
> ;For GRiSP 1 compatibility
> [boot]
> image_path = /media/mmcsd-0-0/rosie_grisp/erts-11.2.2.10/bin/beam.bin
> 
> [erlang]
> args = erl.rtems -C multi_time_warp -- -mode embedded -home . -pa . -root rosie_grisp -boot rosie_grisp/releases/0.1.0/start -internal_epmd epmd_sup -kernel inetrc "./erl_inetrc" -sname rosie_grisp -setcookie MyCookie
> 
> [network]
> ip_self = dhcp
> wlan=enable
> hostname = grisp265
> ```
> 
> Tell me if you need some further infos or logs.

eproxus, 2022-03-09:
@dagyu Good observation. I'll look into updating the instructions and changing so we don't hardcode a `grisp.ini.mustache` there but copy it from the actual source code.

dagyu, 2022-03-09:
Ok I tested and now everything works well, thank you for your support!

dagyu, 2022-03-09:
Ok! Sorry if I closed it.

> Great! I'll keep this issue open until we've fixed the documentation.

eproxus, 2022-03-09:
[image] [Update tutorial about `grisp.ini` networking https://github.com/grisp/grisp/issues/100](https://trello.com/c/V2SJxta1/97-update-tutorial-about-grispini-networking-https-githubcom-grisp-grisp-issues-100)

eproxus, 2022-03-10:
@dagyu I've update the wiki page. Could you take a look and confirm it matches your expectations? https://github.com/grisp/grisp/wiki/Connecting-over-WiFI-and-Ethernet#grisp-ini

dagyu, 2022-03-10:
Yes, I think that now all the configurations steps are well explained

closed 2022-03-10 by eproxus