grisp/grisp #101: Missing grisp.ini args in networking tutorial

ecpeterson, 2022-03-30:
### Versions

n/a, this is an error in the documentation.

### Description

In the tutorial for getting network connectivity off the ground, [this subsection](https://github.com/grisp/grisp/wiki/Connecting-over-WiFI-and-Ethernet#grisp-ini) advises adding the string

```
-sname {{release_name}} -setcookie MyCookie
```

to the `args` field in `grisp.ini`. With this alone, the port mapper daemon does not start, and the VM exits after displaying the error

```
Protocol 'inet_tcp': register/listen error: econnrefused
```

The preloaded demo also comes with a [grisp.ini](https://github.com/grisp/grisp_demo/blob/main/grisp/grisp2/common/deploy/files/grisp.ini.mustache), and its `args` field has some extra flags:

```
-internal_epmd epmd_sup -kernel inetrc "./erl_inetrc"
```

With these also appended to `args`, the tutorial behaves as expected.

peerst, 2022-03-30:
Sigh this was correct before (very sure on that because ran a bunch of tutorials based on what's in the Wiki)  but rewriting for GRiSP2 and Ethernet seems to have removed that part.

Fixed it, thanks!

peerst, 2022-03-30:
Fixed in Wiki

closed 2022-03-30 by peerst

ecpeterson, 2022-03-30:
No, thank you! This project is super cool, and I’m thrilled that it all works as smoothly as it does.