grisp/grisp #102: [QUESTION] When will you add OTP 24 GRISP package support?

aleklisi, 2022-04-09:
How to work with GRISP and OTP 24?

peerst, 2022-04-11:
We discovered a problem with our current state of the OTP patch management system which we tried to improve for GRiSP2 but we are not quite there yet.  So going beyond 23.3.4.11 will be some extra work for us.  Unfortunately we are before very important deadlines in customer projects, so I fear it will take some time (i.e. mid to end of May)

If someone with sufficient rebar3 knowledge wants to work on that externally we can facilitate that though.

Also please note that just specifying OTP version as "23" is broken and versions 23.3.4.12 and higher don't work either:

`{grisp, [{otp, [{version, "= 23.3.4.11"}]}]}. `

Thats how it needs to be configured until we fixed the current problem.

After we fixed the patch system moving towards newer version would be semi automatic until something really incompatible happens.  But we are not quite there yet

aleklisi, 2022-07-18:
I assume that with the 2.2.0 release this issue gets resolved 🎉

closed 2022-07-18 by aleklisi