grisp/grisp #144: Errors when building with `rebar3 grisp build`

odo, 2024-10-20:
### Versions

* OS: MacOS 15.0.1 Sequoia
* GRiSP Toolchain:  grisp2-rtems-toolchain_macOS_11.6.3_3122986b9cd7073f42f1387f3981c812a2909b68
* Erlang: 25.3.2.15
* Rebar: rebar 3.24.0 on Erlang/OTP 25 Erts 13.2.2.11

### Description

I am trying to build with the a pre-compiled toolchain but I get these errors in `grisp/_grisp/grisp2/otp/25.3.2.15/build/erts/config.log`:

```
configure:4340: /Users/odo101/src/GRiSP/grisp2-rtems-toolchain_macOS_11.6.3_3122986b9cd7073f42f1387f3981c812a2909b68//bin/arm-rtems5-gcc --pipe -B /Users/odo101/src/GRiSP/grisp2-rtems-toolchain_macOS_11.6.3_3122986b9cd7073f42f1387f3981c812a2909b68//arm-rtems5/imx7/lib -specs bsp_specs -qrtems -V >&5
arm-rtems5-gcc: error: unrecognized command line option '-V'
arm-rtems5-gcc: fatal error: no input files
compilation terminated.
```

[config.log](https://github.com/user-attachments/files/17450071/config.log)

Any help is appreciated, happy to provide more details if needed.

sylane, 2024-10-21:
Weird this GCC option is deprecated, I'll try to find if it is still used somewhere...

odo, 2024-10-21:
I retried with OTP 27.1.2 but it seems to have the same problem.
[config.log](https://github.com/user-attachments/files/17464062/config.log)

ziopio, 2024-10-22:
@odo please use `grisp 2.3.0` and OTP 25 and try again, that prebuilt toolchain is quite old and misses few libs.
But I cannot reproduce your failure...

@sylane we should mention this in the wiki and deprecate those packages. 

@odo I strongly suggest you to use the docker image toolchain, is up to date.
We generally use docker or build it from scratch as we do not maintain prebuilt toolchain packages anymore.

odo, 2024-10-23:
Thanks for looking into it! In fact I switched to the docker approach using OTP `27.1.2` a few days ago which works really well.
So from my side, this issue can be closed

sylane, 2024-10-23:
I am happy this is working for you now !
I'll close the issue then.

closed 2024-10-23 by sylane