grisp/grisp #156: Deploy pre-built package error with tutorial build

Nicodaube, 2025-01-21:
Hi ! 

I'm trying to go through the tutorials on the wiki but have been getting error when trying to deploy the basic app created in the "Creating Your First GRiSP Application" tuto. I'm currently running ```rebar 3.24.0 on Erlang/OTP 27 Erts 15.2``` . I first tryed to create the app with OTP 25 (basic value given by ```rebar3 grisp configure``` ) but got the same error.  This are the steps I followed after the basic rebar3 grisp configure and modified the code as told in the tutorial.

```rebar3 compile``` 

[image]

[image]

```rebar3 grisp deploy``` 

[image]

And these are my build_overlay and hash given by ```rebar3 grisp report``` 
[build_overlay.txt](https://github.com/user-attachments/files/18492606/build_overlay.txt)

[hash.txt](https://github.com/user-attachments/files/18492608/hash.txt)

I would really appreciate some help, as I don't really see where is my problem.

Thanks a lot

Nicodaube, 2025-01-22:
There you go : [deploy_error.txt](https://github.com/user-attachments/files/18505179/deploy_error.txt)

I'm on Ubuntu 24.04.1

ziopio, 2025-01-22:
I finally catched the problem, you are experiencing this bug because your project is under a path that contains `Mémoire`

Sadly it looks like our plugin is not robust with letters like `é` appearing in paths.

Place the project somewhere else while we cleanup this stupid error, we will open another issue on the correct repository.

closed 2025-01-22 by Nicodaube